cyanfish/naps2 #628: Remove superfluous "Apply to all ... selected images" checkbox

githubuser181226, 2025-05-31:
Hello.

I find the existence of `Apply to all ... seleted images` checkbox inside the image processing window (crop, birghtness/contrast etc.) completely superfluous.

The very action of manually selecting multiple images for processing is enough to implicate a decision to process all the selected images at once.

If I wanted to process just a single image, I'd select just a single image.

Please, consider removing the checkbox in question and only leave a message telling how many images have been selected for processing i.e. "Changes will apply to all XXX selected images."

Thank you.

cyanfish, 2025-06-08:
I'm not going to remove it, but in [8.2.0](https://www.naps2.com/download) it will stay checked in the future if you check it once.

closed 2025-06-08 by cyanfish

githubuser181226, 2025-06-12:
Thank you for considering this and changing the default behavior.
Having this option checked by default resolves my issue.

However, I do wonder what is the real life use case for it not be checked by default ever?

Also, if you select multiple images and issue "rotate right" command, NAPS does not ask for the same confirmation ie. click ok to change all selected images". I find it inconsisstent.

And there is also the question, why, if unchecked, NAPS2 always changes only the first image in the selection and does not allow for changing it to any other, thus duplicating the functionality of issuing a command on a single selected image.

Again, thanks and keep it up.
NAPS even with all its flaws is IMO the best on the market for what it is meant to be.
I tip my hat for you, Sir.