cyanfish/naps2 #648: NAPS2.exe is missing while using scan server

Exlord, 2025-07-12:
I am trying to use the scan server/client. I started with the https://github.com/cyanfish/naps2-webscan.
The original code was working with the default options.
I modified it to be able to get the list of devices to client first and then allow the web client to choose and modify the  scanner and its options.

I setup a `EmbedIO` server inside the worker and tried starting the scan server inside a EmbedIO action (request from web)

Here is all my worker/service code :

``` C#

using EmbedIO;
using EmbedIO.Actions;
using Google.Protobuf.WellKnownTypes;
using Microsoft.Win32;
using NAPS2.Escl;
using NAPS2.Escl.Server;
using NAPS2.Images;
using NAPS2.Images.ImageSharp;
using NAPS2.Remoting.Server;
using NAPS2.Scan;
using NAPS2.Threading;
using NTwain.Data;
using Swan.Logging;

namespace Exlord.WebScan.LocalService;

public class Worker : BackgroundService
{
    private readonly ILogger<Worker> _logger;
    // Our web server is disposable.
    private ScanServer _scanServer;
    CancellationToken _stoppingToken;

    public Worker(ILogger<Worker> logger)
    {
        _logger = logger;
    }

    protected override async Task ExecuteAsync(CancellationToken stoppingToken)
    {
        _stoppingToken = stoppingToken;
        var url = "http://localhost:9881/";
        using (var server = CreateDeviceServer(url))
        {
            Console.WriteLine(" ============ device server created");
            server.Start(stoppingToken);
            Console.WriteLine(" ============ device server started");
            CreateScanServer();
            await stoppingToken.WaitHandle.WaitOneAsync();
            if (_scanServer != null)
            {
                await _scanServer.Stop();
                _scanServer = null;
            }
        }
    }

    // Create and configure our web server.
    private WebServer CreateDeviceServer(string url)
    {
        var server = new WebServer(ws => ws
            .WithUrlPrefix(url)
            .WithMode(HttpListenerMode.EmbedIO))
            .WithCors("*")
            // action to get a list of devices
            .WithModule(new ActionModule("/devices", HttpVerbs.Get, async ctx =>
            {

                using var scanningContext = new ScanningContext(new ImageSharpImageContext());
                scanningContext.SetUpWin32Worker();
                var controller = new ScanController(scanningContext);

                var wiaDevices = await controller.GetDeviceList(Driver.Wia);
                var twainDevices = await controller.GetDeviceList(Driver.Twain);

                await ctx.SendDataAsync(wiaDevices.Concat(twainDevices).ToArray());
            }))
            // action to save selected device 
            .WithModule(new ActionModule("/devices", HttpVerbs.Put, async ctx =>
            {
                Console.WriteLine(" ======= got new device");
                Globals.SelectedDevice = null;
                ScannerDevice device = await ctx.GetRequestDataAsync<ScannerDevice>();

                Console.WriteLine($" ============= saving device {device.Name}");

                RegistryKey key = Registry.LocalMachine.OpenSubKey("Software", true);
                RegistryKey subKey = key.CreateSubKey(Consts.RegKey, true);
                subKey.SetValue(Consts.RegKeyDeviceDriver, device.Driver);
                subKey.SetValue(Consts.RegKeyDeviceId, device.ID);
                subKey.Close();
                key.Close();

                // stop the previes device sharing server
                if (_scanServer != null) await _scanServer.Stop();

                // strat the sharing server with the new selected device
                CreateScanServer();
            }));

        // Listen for state changes.
        server.StateChanged += (s, e) => $"WebServer New State - {e.NewState}".Info();

        return server;
    }

    private async void CreateScanServer()
    {
        Console.WriteLine(" ======== creating scan server");
        var device = Globals.GetSelectedScanner();
        if (device == null)
        {
            Console.WriteLine(" ============= no device is selected");
            return;
        }

        using var scanningContext = new ScanningContext(new ImageSharpImageContext());
        scanningContext.Logger = _logger;
        scanningContext.SetUpWin32Worker();

        using var scanServer = new ScanServer(scanningContext, new EsclServer
        {
            // This line is required for scanning from a browser to work
            SecurityPolicy = EsclSecurityPolicy.ServerAllowAnyOrigin,
        });

        var controller = new ScanController(scanningContext);
        var devices = await controller.GetDeviceList(new ScanOptions
        {
            Driver = device.Driver.ConvertToEnum<Driver>(),
            TwainOptions = new TwainOptions
            {
                Dsm = TwainDsm.NewX64,               
            },
            //PaperSource = PaperSource.Feeder,
            //PageSize = PageSize.A4,
            //Brightness
            //Contrast
            //Dpi
            //MaxQuality
            //Quality            
        });
        var firstDevice = devices.Find(d => d.ID == device.ID);
        if (firstDevice == null)
        {
            Console.WriteLine(" ========== selected device not found");
            return;
        }
        
        Console.WriteLine($" ========== registered '${firstDevice.Name}' on port 9880");
        scanServer.RegisterDevice(firstDevice, port: 9880);

        // Share the device(s) until the service is stopped
        Console.WriteLine(" ========== starting the server");
        await scanServer.Start();
        Console.WriteLine(" ========== server started");

        _scanServer = scanServer;
    }
}

```

I am getting this error :

```
Exlord.WebScan.LocalService.Worker[0]
      Could not start worker
      System.ComponentModel.Win32Exception (2): An error occurred trying to start process 'E:\exlord-WebScanner\Exlord.WebScan.LocalService\bin\Debug\net8.0-windows\win-x64\NAPS2.exe' with working directory 'E:\exlord-WebScanner\Exlord.WebScan.LocalService'. The system cannot find the file specified.
         at System.Diagnostics.Process.StartWithCreateProcess(ProcessStartInfo startInfo)
         at System.Diagnostics.Process.Start(ProcessStartInfo startInfo)
         at NAPS2.Remoting.Worker.WorkerFactory.StartWorkerProcess(WorkerType workerType)
         at NAPS2.Remoting.Worker.WorkerFactory.<>c__DisplayClass14_0.<StartWorkerService>b__0()
```

I have installed everything throw nuget...
I have installed the `NAPS2.Sdk.Worker.Win32` package and have the `NAPS2.Worker.exe` but no `NAPS2.exe` .
So where and how can I get the NAPS2.exe? Or am I doing something wrong ?

Exlord, 2025-07-13:
It works with the original code but not with mine. The server does not start and the scan request returns `ERR_CONNECTION_REFUSED`

cyanfish, 2025-07-13:
You might need to add some more error handling to catch the issue. But one thing that stands out to me is that the service might not have permission to write to the registry.

Exlord, 2025-07-13:
I have added app manifest for registery permission and it asks to run VS as admin and it works just fine
`<requestedExecutionLevel level="requireAdministrator" uiAccess="false" />`
Ok lets say I catched the error what next? how to exactly handle it ?

Exlord, 2025-07-21:
It seems the error message shown above is not the problem, the original `https://github.com/cyanfish/naps2-webscan` also is throwing this error when logging is enabled ... but it keeps doing its job and its correct ...

My problem seems to be with the server not starting ...

I simplified the start server command and actually used the obne from the sample to see if my code was the issue 

```C#
            .WithModule(new ActionModule("/start", HttpVerbs.Post, async ctx =>
            {
                Console.WriteLine("============== Starting The Server ...");

                using var scanningContext = new ScanningContext(new ImageSharpImageContext());
                scanningContext.Logger = _logger;
                scanningContext.SetUpWin32Worker();               
                var controller = new ScanController(scanningContext);

                var devices = await controller.GetDeviceList(Driver.Twain);
                var firstDevice = devices.First();

                using var scanServer = new ScanServer(scanningContext, new EsclServer
                {
                    // This line is required for scanning from a browser to work
                    SecurityPolicy = EsclSecurityPolicy.ServerAllowAnyOrigin,
                    Logger = _logger,                   
                });

                // You'll need to manually specify a port as device discovery doesn't work from a browser.
                // If you're sharing multiple devices make sure to give them each a unique port number.
                scanServer.RegisterDevice(firstDevice, port: 9880);

                // Share the device(s) until the service is stopped
                await scanServer.Start();

                _scanServer = scanServer;

                await ctx.SendStringAsync("Done", "Text", System.Text.Encoding.UTF8);
            }));
```

this code is not throwing any errors related to the server ... but I am not getting any logs from it and if I try to run the scan command from the frontend it says : 

```
scanner err AxiosError {message: 'Network Error', name: 'AxiosError', code: 'ERR_NETWORK', config: {…}, request: XMLHttpRequest, …}

GET http://localhost:9880/eSCL/ScannerCapabilities net::ERR_CONNECTION_REFUSED
```

So how can I enable the logs for the server ?
Why is it not running ?

cyanfish, 2025-07-21:
The problem might be that you're creating multiple scan servers in succession which leaves the port 9880 in use. If the port is already used NAPS2 will pick another port at random. So you can try making sure you don't register a device on the port more than once, or maybe a random port and share that with the client/server.

cyanfish, 2025-07-21:
Or another possibility is that the client is simply trying to connect too quickly and not giving the server time to start listening on the port.

closed 2025-08-04 by cyanfish