cyanfish/naps2 #649: Windows ARM64 build unable to use TWAIN drivers

squarestopsthere, 2025-07-12:
**Describe the bug**
I'm running Windows 11 on a computer with a Snapdragon X Elite CPU. Running the ARM64 build of NAPS2, I'm unable to use the TWAIN drivers, which is an issue because my scanner's WIA drivers don't work on here. I can't see the TWAIN driver in the NAPS2 GUI and, when running `NAPS2.Console.exe" --listdevices --driver twain` returns `The "Twain" driver is not supported on this platform.`.

The scanner is the [IRIScan Anywhere 6 Wifi Duplex](https://www.irislink.com/EN-US/c2466/iriscan-anywhere-duplex).

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 26200.5670

cyanfish, 2025-07-12:
Does the TWAIN driver actually work on arm64? If you're not sure you can try the [portable x64 zip](https://github.com/cyanfish/naps2/releases/download/v8.2.0/naps2-8.2.0-win-x64.zip) which should run. But I excluded TWAIN from the arm64 build because as far as I know it shouldn't generally work on arm64 (though I could be wrong).

squarestopsthere, 2025-07-12:
Yes, I've been using the TWAIN driver on my ARM64 computer for a good while. I spent all day yesterday troubleshooting the issue after reinstalling NAPS2 from winget and not realising the switch from x64 to arm64 builds. `NAPS2.Console --listdevices --driver twain` with the x64 build lists the scanner I own. I haven't installed my second scanner but was using it previously with NAPS2 on this same computer also with the x64 build.

cyanfish, 2025-08-31:
This should be fixed in [8.2.1](https://www.naps2.com/download).

closed 2025-08-31 by cyanfish