cyanfish/naps2 #652: Cannot save PDF settings after enabling encryption

bgoodmansf, 2025-07-14:
**Describe the bug**
After checking the "Encrypt PDF" checkbox in the PDF settings dialog, it is not possible to uncheck the "Encrypt PDF" checkbox and save the settings. The "Remember these settings", "Restore Defaults", "OK" and "Cancel" controls all disappear. Therefore it is not possible to click "OK" to save the change. This also means that it is not possible to restore defaults after Encrypt PDF has been checked.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a scan.
2. Open the PDF settings.
3. Check "Encrypt PDF".
4. Click OK to save the settings.
5. Open the PDF settings.
6. Uncheck "Encrypt PDF".

**Expected behavior**
OK button should be visible.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Fedora 42 with KDE desktop
 - Version 8.2.0.0

cyanfish, 2025-08-31:
This should be (mostly) fixed in [8.2.1](https://www.naps2.com/download).

closed 2025-08-31 by cyanfish

bgoodmansf, 2025-08-31:
Looks good. Thanks!