cyanfish/naps2 #669: Missing Google/Microsoft e-mail provider in the Linux version

miku-orb, 2025-08-01:
**Describe the bug**
The Windows version has an option to send e-mails via Gmail or Microsoft via the built-in providers that do OAuth stuff. It seems that they're both missing in the Linux version.

According to the code, it seems they should be shown, as long as the client credentials are there (`default.credentials.json` file that should be embedded somewhere? seems to be missing).

**To Reproduce**
1. Install Linux version (rpm, but deb seems to be the same, didn't test flatpak)
2. Open list of e-mail providers
3. You see only `Thunderbird` (grayed out or not, depending on whether it's detected – it doesn't detect Flatpak version, but that's a separate bug)

**Expected behavior**
It should be visible and usable just like on Windows (since it's neat).

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Fedora (rpm) or Arch Linux (naps2-bin repacked from rpm)
 - Version [e.g. 7.1.0] 8.2.0.0

**Additional context**
Assuming the browser callback thing would work and that it wasn't disabled because of it... (including that either NAPS2 or browser can be a Flatpak or not, that's 4 combinations to test...)

cyanfish, 2025-08-02:
Yeah that's my bad, I switched Linux build systems and didn't properly copy the creds over. I'll fix it in the next version.

cyanfish, 2025-08-31:
This should be fixed in [8.2.1](https://www.naps2.com/download).

closed 2025-08-31 by cyanfish