cyanfish/naps2 #681: Not really a bug: scanner blocks when changing from ADF to flatbed

DrMartinus, 2025-08-26:
I use a brother MFC-J6920DW, a workhorse that has done me very good service and continues to do so, and my OS is Manjaro (Arch Linux).
Before I used naps2, I used another software to create pdf documents using eiter ADF or flatbed scanner, whatever was appropriate (you can't put a book or magazine in the ADF, and some papers are too thin to be run through the ADF etc.). I noticed then that, in an irregular interval, the scanner refused to scan from flatbed after it had scanned a document using the ADF. This didn't happen always, sometimes the change went smoothly. 
Now using naps2, when I change from one profile (ADF...) to another (Flatbed), at the first scan attempt, the scanner takes a while, then gives a beep and an error pops up that communication to the scanner was interrupted (or similar).
It takes a while (5 to 10 seconds at least) and becomes annoying, because it really interrupts the workflow. I'm sure this is an issue with the scanner, yet I wonder if something can be done on the software side, because with the software I used before, this problem didn't occur always, but only randomly, maybe every third or fourth change from ADF to flatbed.
Any ideas?

alicavus, 2025-09-10:
Debug information is from different time related to error log.
Error log is from 2025-09-02:
```
 2025-09-02 17:37:22.8510 5986 Killing unresponsive worker
```
while debug information date is 2025-09-09.

cyanfish, 2025-09-10:
That error is unlikely to be related. It looks like you're scanning with SANE. That device does support ESCL, so if you have it connected to your network, you can try scanning with ESCL Driver instead.

DrMartinus, 2025-12-27:
Sorry for the very late reply. In Linux, it seems there is no way to install an ESCL driver. There is only a kind of ESCL supplement, that allows to address the scanner via the network. I gave it a try, but I gave up on that one.
It solved the issue I was talking about initially, i.e. no complaint about connecting to the scanner. But it takes quite long to connect via the network, and then it always moves the scan head across the entire scannable area, even if I (what is usually the case) have only marked a small portion of it to be scanned. The scanned image then is only the marked section, so that is ok, but the scan head takes more time to scan than when using the SANE driver. And since this happens with every scan, it adds up. Hence I went back to the SANE driver entirely and acknowledge the error message.