cyanfish/naps2 #686: Scan and direct Upload to URL / API and online services

x9t9, 2025-08-31:
I apologize in advance if I missed something and a feature like that already exists. 
I made some search and could not find it, if it is there- please point me in the right direction.

**Is your feature request related to a problem? Please describe.**
**_TL,DR;_** 
Automate the integration of 3rd party online services API's via a simple URL

**_Detail_** 
We all know the countless systems, services and online apps, both self-hosted or not, to store, archive, manage all types of documents and images / Photos.
The process of scanning to local File system, Save , Rename if needed - and then manually upload to any of those systems can be very laborious task.
The majority of these services have some kind of simple REST API to allow integration with other systems apps or services.

**Describe the solution you'd like**
A simple "Send to URL" or API would allow users to automatically send the scanned doc to any of these services.
Of course there are hundreds of services, and not all need to be integrated specifically, but a simple UI to input a "Target" URL and credentials would allow each user to configure its own preferred service

**Describe alternatives you've considered**
Some custom niche Drivers that would allow expanding with plugins in other languages to create my own workflow, or just automate uploads to FTP which requires additional steps.
This is not applicable to most users.

**Additional context**
DMS ( Document management ) Services, Image management and hosting Services, Personal Clouds, Workflow Processors, Archives, and countless other examples of online services.

cyanfish, 2025-08-31:
#106 is essentially the same idea

daVe55w, 2026-02-04:
IMO, this would be a very useful option. The more popular online storage sites like Dropbox, Box, Google Drive, and OneDrive could be included, along with the option to enter a specific URL for another cloud storage site. Brother's iPrint&Scan software has a Cloud Storage section like that. When the user clicks one of the cloud storage sites, it opens that site's login page in the browser. The user then logs in to the site and iPrint&Scan is authenticated. If the user wants to upload a scan to a particular cloud service, he/she simply clicks that service's icon. Here's a screenshot of what it looks like.

[image]