cyanfish/naps2 #687: Feature Request: Optional AI‑Accelerated OCR with Interactive Preview, Correction, and Incremental Learning

MrGrymReaper, 2025-08-31:
**Is your feature request related to a problem? Please describe.**
Yes. NAPS2’s current OCR functionality is based on Tesseract, which is CPU-bound and lacks hardware acceleration. It can be slow or inaccurate on complex scans, handwritten text, or noisy documents. Additionally, users cannot preview or correct OCR results before saving, which limits control and confidence in the output.

**Describe the solution you'd like**
Introduce an optional AI-powered OCR mode that leverages modern deep learning models and hardware acceleration when available. This mode would:

-     Use a hardware-agnostic inference runtime (e.g. ONNX Runtime or Apache TVM) to support acceleration across:

         -   NVIDIA GPUs (Tensor Cores via CUDA/TensorRT)

         -   AMD GPUs (via ROCm)

         -   Intel NPUs (via OpenVINO)

         -   Apple Silicon NPUs (via Core ML)

         -   Arm NPUs (via Arm NN / Ethos)

         -   DirectML/OpenCL as universal GPU fallback

-     Provide a live OCR preview window:

         -   Split view: scanned image on the left, editable recognised text on the right

         -   Highlight low-confidence words (e.g. yellow/red) based on model confidence scores

         -   Allow inline editing before saving

-     Support incremental learning:

         -   Store user corrections locally as (image snippet → corrected text) pairs

         -   Periodically fine-tune the OCR model using lightweight methods (e.g. LoRA adapters)

         -   All training remains on-device and privacy-preserving

-     Include fallback logic:

         -   If hardware is unavailable, prerequisites are missing, or the user disables AI OCR (globally or per scan), NAPS2 falls back to the current Tesseract-based OCR

**Describe alternatives you've considered**

- Continuing with Tesseract alone: reliable but limited in accuracy and speed, especially on modern hardware

- External OCR tools: break NAPS2’s offline, integrated workflow

- Post-processing in other apps: adds friction and reduces user control

**Additional context**
This feature would be opt-in and non-disruptive. It improves performance and accuracy for users with supported hardware, while preserving the current experience for everyone else.

Mock-ups and diagrams available to illustrate the proposed UI and workflow:

- [image]

- [image]

This proposal is designed to be modular, hardware-independent, and user-centric — aligning with NAPS2’s goals of simplicity, privacy, and flexibility.

MrGrymReaper, 2025-08-31:
Tagging @cyanfish for visibility, as well as @SHJordan, @neo269, and @n00ris who’ve been active in recent OCR discussions.

I’d love your thoughts on whether this proposed AI‑powered OCR mode — with hardware‑agnostic acceleration, an interactive preview/correction stage, and a fallback to Tesseract — could fit well within NAPS2’s architecture.

The goal is to keep it opt‑in and non‑disruptive, while offering significant speed and accuracy improvements for users with supported hardware.

Full mock‑ups and workflow diagrams are included in the issue for reference. Feedback on feasibility, UI integration, or potential pitfalls would be hugely appreciated.

cyanfish, 2025-08-31:
Is there an existing open source OCR system that implements these kinds of models and has a compatible license? NAPS2 as a project is a scanning interface, not an OCR engine.

OCR UI improvements are possible (and planned) and I'm willing to consider alternatives to Tesseract, but creating models from scratch is outside of this project's scope.

MrGrymReaper, 2025-08-31:
I have an idea for the architectural structure for the FR.

| **Layer**             | **Role**                                      | **Hardware Independence Strategy**                                                                 |
|-----------------------|-----------------------------------------------|-----------------------------------------------------------------------------------------------------|
| **OCR Model**         | Text detection + recognition (e.g., PaddleOCR, docTR, TrOCR) | Choose a model in PyTorch or TensorFlow so you can export to ONNX                                  |
| **Intermediate Format** | ONNX                                          | Acts as a universal bridge between frameworks and runtimes                                         |
| **Inference Runtime** | ONNX Runtime / TVM / OpenVINO                 | Supports multiple execution providers for GPUs & NPUs                                               |
| **Hardware Backends** | DirectML, Core ML, OpenCL, ROCm, CUDA, SYCL   | Selected at runtime based on available hardware                                                     |

MrGrymReaper, 2025-08-31:
OCR Libraries - Traditional & AI Enhanced:

| **Library / Model** | **Highlights**                                                                 | **Hardware Acceleration**                                                                                   | **License**   |
|---------------------|----------------------------------------------------------------------------------|--------------------------------------------------------------------------------------------------------------|---------------|
| **Tesseract OCR**   | Classic, battle-tested OCR engine; supports 100+ languages; best for clean printed text | Primarily CPU; can be compiled with OpenCL for some GPU acceleration, but no native NPU/Tensor Core support | Apache 2.0    |
| **EasyOCR**         | PyTorch-based, 80+ languages, simple API; good for quick integration            | Native GPU acceleration (CUDA for Tensor Cores); can run on Jetson devices; experimental NPU support via PyTorch backends | Apache 2.0    |
| **PaddleOCR**       | High-accuracy PP-OCRv4 pipeline; multilingual; layout/table/formula recognition | GPU acceleration (CUDA/TensorRT); ARM/edge builds; supports ONNX export for hardware-agnostic deployment     | Apache 2.0    |

MrGrymReaper, 2025-08-31:
@cyanfish 
Part 1: ONNX Runtime in C# (.NET)

ONNX Runtime is a high-performance inference engine that supports a wide range of hardware backends and is fully compatible with C# via NuGet packages. It’s the key to unlocking hardware independence in NAPS2.
✅ Setup in C#

You can integrate ONNX Runtime into NAPS2 using:

```
dotnet add package Microsoft.ML.OnnxRuntime
dotnet add package System.Numerics.Tensors
```

Then in C#:

using Microsoft.ML.OnnxRuntime;
using Microsoft.ML.OnnxRuntime.Tensors;

// Load model
using var session = new InferenceSession("model.onnx");

// Prepare input tensor
var inputData = new DenseTensor<float>(new[] { 1, 3, 224, 224 }); // example shape
var inputs = new List<NamedOnnxValue> {
    NamedOnnxValue.CreateFromTensor("input", inputData)
};

// Run inference
using var results = session.Run(inputs);

✅ Supported Hardware Backends

| **Backend**               | **Supported via ONNX Runtime**              |
|---------------------------|---------------------------------------------|
| **CPU**                   | ✅ Default execution provider                |
| **NVIDIA CUDA / TensorRT**| ✅ With GPU-enabled build                    |
| **DirectML (Windows GPU)**| ✅ For AMD/Intel/NVIDIA GPUs                 |
| **Core ML (Apple Silicon)**| ✅ macOS/iOS support                        |
| **OpenVINO (Intel VPU)**  | ✅ For Intel NPUs                           |
| **ROCm / OpenCL**         | ✅ With custom builds                        |

You can dynamically select the best available provider at runtime.

🔀 Part 2: Hybrid Integration Strategy for NAPS2

Here’s a modular strategy that combines PaddleSharp and ONNX Runtime to give NAPS2 both native .NET performance and hardware flexibility:

🧩 Architecture Overview

| **Layer**              | **Primary Tool**         | **Fallback / Extension**                          |
|------------------------|--------------------------|---------------------------------------------------|
| **OCR Engine**         | PaddleSharp (C#)         | ONNX-exported PaddleOCR / TrOCR                   |
| **Inference Runtime**  | PaddleSharp native       | ONNX Runtime (C#)                                 |
| **Hardware Acceleration** | CUDA via PaddleSharp     | DirectML, Core ML, OpenVINO via ONNX Runtime      |

🔄 Workflow

1.     Default Path:

     -    Use PaddleSharp for fast, native OCR on Windows/Linux with CPU or CUDA GPU.

2.     Fallback Path:

    -    If PaddleSharp is unavailable or unsupported (e.g. on macOS, ARM, NPU), switch to ONNX Runtime.

    -    Load ONNX-exported model (e.g. TrOCR or PaddleOCR) and run inference using the best available provider.

3.     Preview UI:

    -    Unified OCR preview window regardless of backend.

    -    Highlight low-confidence words using model output scores.

     -    Allow inline correction before saving.

4.     Settings Panel:

    -    Let users choose between “Fast (PaddleSharp)” and “Flexible (ONNX)” modes.

    -    Auto-detect hardware and suggest optimal path.

🧠 Benefits

-     Performance: PaddleSharp gives fast startup and low latency on supported systems.

-     Portability: ONNX Runtime unlocks macOS, ARM, and NPU support.

-     Modularity: Easy to extend with new models or runtimes.

-     User Control: Opt-in, non-disruptive, and privacy-preserving.

MrGrymReaper, 2025-08-31:
Visual diagram of how it can be integrated.

[image]

MrGrymReaper, 2025-08-31:
✅ Notes:

    Replace "input" with the actual input node name from your ONNX model.

    You can inspect model metadata using session.InputMetadata and session.OutputMetadata.

    For image-based OCR models, you’ll need to preprocess the image (resize, normalize, convert to tensor) before inference.

⚠️ Common Pitfalls to Avoid

| **Issue**                 | **Fix**                                                                                   |
|---------------------------|--------------------------------------------------------------------------------------------|
| **Input shape mismatch**  | Use [Netron](https://netron.app) or `session.InputMetadata` to inspect expected input shape |
| **Missing native binaries** | Ensure correct platform-specific NuGet package is installed (e.g. CUDA, DirectML, OpenVINO) |
| **Incorrect tensor type** | Use `DenseTensor<float>` or `DenseTensor<byte>` depending on model input requirements     |
| **GPU not used**          | Confirm execution provider is set correctly (e.g. CUDA, DirectML) and GPU drivers are installed |

MrGrymReaper, 2025-08-31:
@cyanfish Use https://github.com/nietras/OnnxSharp to parse and manipulate, to inspect the models. To ensure that the expected input shape is right.

MrGrymReaper, 2025-08-31:
@cyanfish Additional Note – Handling Visual “Noise” in OCR
In many real‑world scans, text is accompanied or obscured by non‑text elements that can reduce OCR accuracy. Common examples include:
Images near text – graphics or icons adjacent to text can confuse segmentation, leading to partial or merged word detection.
Watermarks behind text – semi‑transparent logos or patterns alter glyph shapes and pixel intensity, lowering recognition confidence.
Handwriting over printed text – overlapping strokes distort printed characters, causing misclassification or extra “phantom” text.
To improve robustness, the proposed AI‑powered OCR mode could:
Apply pre‑processing filters (background removal, binarisation, watermark suppression).
Use layout analysis to separate text from nearby images.
Leverage confidence scoring to flag low‑certainty regions for user review in the preview/correction stage.
These enhancements would help ensure that noisy documents are still processed accurately, while keeping the workflow opt‑in and non‑disruptive.

MrGrymReaper, 2025-08-31:
@cyanfish Check out the workflow text document it shows how it would work. Namely TesseractOCR isn’t going away completely, just becoming a fallback, with AI OCR being the primary.