cyanfish/naps2 #689: The 'link' start tag does not match the end tag of 'head'

MattDahEpic, 2025-09-01:
I've updated to version 8.2.1 which contains the fix for #654 and am now running into a new error message:

[image]

<details>
<summary>Full stack trace</summary>

```
System.Xml.XmlException: The 'link' start tag on line 18 position 6 does not match the end tag of 'head'. Line 34, position 15.
   at void System.Xml.XmlTextReaderImpl.Throw(Exception e) x 2
   at void System.Xml.XmlTextReaderImpl.ThrowTagMismatch(NodeData startTag)
   at void System.Xml.XmlTextReaderImpl.ParseEndElement()
   at bool System.Xml.XmlTextReaderImpl.ParseElementContent()
   at void System.Xml.Linq.XContainer.ReadContentFrom(XmlReader r) x 2
   at XDocument System.Xml.Linq.XDocument.Load(XmlReader reader, LoadOptions options)
   at XDocument System.Xml.Linq.XDocument.Parse(string text, LoadOptions options) x 2
   at async Task<XDocument> NAPS2.Escl.Client.EsclClient.DoRequest(string endpoint)
   at async Task<EsclCapabilities> NAPS2.Escl.Client.EsclClient.GetCapabilities()
   at async Task<ValueTuple<EsclClient, EsclCapabilities>> NAPS2.Scan.Internal.Escl.EsclScanDriver.GetEsclClientWithCaps(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents)
   at async Task NAPS2.Scan.Internal.Escl.EsclScanDriver.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<IMemoryImage> callback)
   at async Task NAPS2.Scan.Internal.RemoteScanController.Scan(ScanOptions options, CancellationToken cancelToken, IScanEvents scanEvents, Action<ProcessedImage, PostProcessingContext> callback)
   at async Task NAPS2.Scan.ScanController+<>c__DisplayClass15_1.<Scan>g__DoScan|7(?)+DoScan(?)
   at async IAsyncEnumerable<ProcessedImage> NAPS2.Scan.ScanController.Scan(ScanOptions options, CancellationToken cancelToken)+(?) => { }
```

</details>

This error occurs regardless of which printer I am scanning from. I have an Epson MFC-L2750DW and a Brother EcoTank ET-8550.
The error occurs as soon as I press scan, the printer does not make any noise.
Both printers are using the ESCL driver.
Both printer's errors have the same line numbers and character positions.
Other scanner apps such as the "built-in" GNOME `simple-scanner` work correctly with both printers.
I am running PopOS 22.04.

Please let me know if there's any more info I can provide. I don't see a way to save out the document it's actually trying to parse as I can only guess this and #654 won't be the only parse errors.

Thank you so much for your work on NAPS2! It's my favorite scanning app by far.

cyanfish, 2025-09-01:
Have you tried using the SANE driver? You can also collect debug logs https://www.naps2.com/doc/troubleshooting#debug-log

MattDahEpic, 2025-09-20:
I am able to scan successfully when using the SANE driver, though it takes several minutes to find the printer. This printer used to work with the ESCL driver a long time ago (v7 I think) on Windows but I no longer have a Windows machine to test with to confirm. I'm able to scan again but I'm happy to help investigate further if you are interested in resolving whatever is wrong with the ESCL.